RfidResearchGroup/ChameleonUltra #219: about sleep time bug

juminut, 2024-07-09:
When sleeping, use the button to wake up. Continuously pressing the button will not keep the device awake, and sometimes it will also enter sleep mode. It will not remain awake due to continuous pressing of the button.

Also, please tell me, where in the source code is the control for how long to sleep?
Is it "SLEEP-DELAY-MS-BUTTON-CLICK" in the "/firmware/application/src/utils/syssleep. h" file? I modifying this did not extend the wake-up time.

GameTec-live, 2024-07-09:
Of course long pressing the button doesn't keep it awake? It triggers a function which you can define in the settings.
This also isn't a bug, the CU is designed to wake up when brought into a field or when a button is pressed and then immediately go back to sleep to preserve battery.

I believe the setting is somewhere in those files though.

juminut, 2024-07-10:
sorry,this isn`t a bug.I only connected a 3.3V power supply. No battery connected. The situation that occurred

closed 2024-07-10 by juminut