RfidResearchGroup/ChameleonUltra #223: [idea] add Pyramid card support (here is a code from PM3 repo for it)

Spider624, 2024-08-09:
Pyramid card is Wiegand communication protocol and it is already realized in Proxmark3 repo, so idea is add this code to Chameleon Ultra.
Here is file in proxmark repo - "wiegand.c" (full link is https://github.com/RfidResearchGroup/proxmark3/blob/master/armsrc/wiegand.c)

And there is code for pyramid cards support: 

        case 39 :               // 39bit KERI System Pyramid
            fc &= 0x1FFFF;      // 17bits
            cn &= 0xFFFFFFFF;   // 20bits
            value = fc << 20 | cn;
            num_to_bytebits(value, 37, temp);
            wiegand_add_parity(temp, dest, 37);
            break;
        case 44 :               // 44bit KERI system Pyramid
            oem &= 0xFF;        // 8bits
            fc &= 0xFFF;        // 12bits
            cn &= 0xFFFFFFFF;   // 21bits
            value = oem << 20 | fc << 12 | cn;
            num_to_bytebits(value, 42, temp);
            wiegand_add_parity(temp, dest, 42);
            break;

GameTec-live, 2024-08-09:
if it only was that easy lol

GameTec-live, 2024-08-09:
also #185

Spider624, 2024-08-09:
so sad.
Hope, that it will be added one day

closed 2025-04-05 by GameTec-live