RfidResearchGroup/ChameleonUltra #224: CLI

Waiker, 2024-08-09:
**Describe the bug**
error while running cmake compilation

**To Reproduce**
 WSL2 Ubuntu 20.04
0. `apt-get update`
1. Execute `sudo apt install git cmake build-essential python3-venv`
2. run 
```
#!/bin/bash

git clone https://github.com/RfidResearchGroup/ChameleonUltra.git
(
  cd ChameleonUltra/software/src
  mkdir -p out
  (
    cd out
    cmake ..
    cmake --build . --config Release
  )
)
(
  cd ChameleonUltra/software/script
  python3 -m venv venv
  source venv/bin/activate
  pip3 install -r requirements.txt
  deactivate
)
```
3. Error
```
 waik@Computer:~$ sudo bash setup_chameleon.sh
[sudo] password for waik:
fatal: destination path 'ChameleonUltra' already exists and is not an empty directory.
-- Run on linux.
-- Configuring done
-- Generating done
-- Build files have been written to: /home/waik/ChameleonUltra/software/src/out
[ 15%] Built target mfkey64
[ 31%] Built target mfkey32v2
[ 46%] Built target mfkey32
[ 64%] Built target darkside
[ 66%] Linking C executable /home/waik/ChameleonUltra/software/script/bin/staticnested
/usr/bin/ld: CMakeFiles/staticnested.dir/nested_util.c.o: in function `nested':
nested_util.c:(.text+0x68f): undefined reference to `pthread_create'
/usr/bin/ld: nested_util.c:(.text+0x6d0): undefined reference to `pthread_join'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/staticnested.dir/build.make:174: /home/waik/ChameleonUltra/software/script/bin/staticnested] Error 1
make[1]: *** [CMakeFiles/Makefile2:194: CMakeFiles/staticnested.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
Requirement already satisfied: pyserial==3.5 in ./venv/lib/python3.8/site-packages (from -r requirements.txt (line 1)) (3.5)
Requirement already satisfied: colorama==0.4.6 in ./venv/lib/python3.8/site-packages (from -r requirements.txt (line 2)) (0.4.6)
Requirement already satisfied: prompt-toolkit==3.0.39 in ./venv/lib/python3.8/site-packages (from -r requirements.txt (line 3)) (3.0.39)
Requirement already satisfied: wcwidth in ./venv/lib/python3.8/site-packages (from prompt-toolkit==3.0.39->-r requirements.txt (line 3)) (0.2.13)
```

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
[image]

**Host (please complete the following information):**
 - WSL2 Ubuntu 20.04
 
**Additional context**
Please help with installation

GameTec-live, 2024-08-09:
it says right in your error:
destination path 'ChameleonUltra' already exists and is not an empty directory.

GameTec-live, 2024-08-09:
it not finding pthread is a strange error though? maybe googling will find smthng

Waiker, 2024-08-10:
I solved the problem by using a version of the compiler suitable for my version of the operating system `LDFLAGS=-pthread cmake -DCMAKE_C_COMPILER=gcc-10 ..`

closed 2024-08-10 by Waiker