RfidResearchGroup/ChameleonUltra #225: CLI nested attack error

Waiker, 2024-08-09:
I use CLI for Chameleon. But it does not see the key (when connected to Android, it successfully sees the key).

I did not find any documentation on CLI. Tell me, what is wrong? How to perform an attack and how do I scan the key?

`
[USB] chameleon --> hf mf nested --blk 0 -A -k FFFFFFFFFFFF --tblk 4 --tA
 - Nested recover one key running...
 - NT vulnerable: HardNested
 [!] HardNested has not been implemented yet.
No key found, you can retry.
[USB] chameleon -->`

Foxushka, 2024-08-10:
> NT vulnerable: HardNested
As it says your card is not vulnerable to default nested attack and requires hard nested attack, which isn't implemented at this moment on CU. If you have proxmark3/Flipper Zero you can run attack from them and use recovered keys to read card

Waiker, 2024-08-11:
> > NT vulnerable: HardNested
> 
> As it says your card is not vulnerable to default nested attack and requires hard nested attack, which isn't implemented at this moment on CU. If you have proxmark3/Flipper Zero you can run attack from them and use recovered keys to read card

the documentation supports this type of attack
[image]

Foxushka, 2024-08-11:
> > > NT vulnerable: HardNested
> > 
> > 
> > As it says your card is not vulnerable to default nested attack and requires hard nested attack, which isn't implemented at this moment on CU. If you have proxmark3/Flipper Zero you can run attack from them and use recovered keys to read card
> 
> the documentation supports this type of attack ![Снимок экрана 2024-08-12 020304](https://private-user-images.githubusercontent.com/5464086/356905661-6dbb36e1-0db0-4431-90ac-b5ca793429bd.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjM0MTE1NDQsIm5iZiI6MTcyMzQxMTI0NCwicGF0aCI6Ii81NDY0MDg2LzM1NjkwNTY2MS02ZGJiMzZlMS0wZGIwLTQ0MzEtOTBhYy1iNWNhNzkzNDI5YmQucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDgxMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA4MTFUMjEyMDQ0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YzhkYjEwZGZjYmQzNDdjOTk0YmY3MjA4NGJkOTRkZjRkYjE1ZWVlMDczZjVjMzhmOTY0ZDM2ZWM2N2Y0OTJjZCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.5G5DXpV6uuw2e9HoQ5XzKw5O0PlyllXnapdGI_7QuF0)

Application layer: not yet supported

Waiker, 2024-08-11:
> Уровень приложений: пока не реализован

but, **Does the software support - suuport** ?

GameTec-live, 2024-08-12:
That probably just means SDK or similar, but if the application doesn't support it, it's not there.

closed 2025-04-05 by GameTec-live