RfidResearchGroup/ChameleonUltra #226: Does not go to sleep

Kabron287, 2024-08-11:
**Describe the bug**
After some playng with Bluetooth connection my device turn permanently LED1, does not react on buttons and wont go to sleep.
Bluetooth also gone. 
But PC connection works as expected.

Medicine: Factory Reset.  

**To Reproduce**
Some manipylations with BLE connection.

**Host (please complete the following information):**
FW: v2.0.0-200-gc771ae9

mitmarcus, 2024-08-14:
Can confirm acts very strangely, cant upload any card to slot 4

GameTec-live, 2024-08-15:
@mitmarcus you raised your own issue on that topic that seems unrelated with this issue

mitmarcus, 2024-08-15:
> @mitmarcus you raised your own issue on that topic that seems unrelated with this issue

correct, could be two separate issues