RfidResearchGroup/ChameleonUltra #235: Chameleon ultra doesn't works

birmanish, 2024-09-26:
Hello, I just receive my chameleon ultra from mtools tech but it don't want to recognize my lf badge (rfid). I tried with another mifare classic 1k card but not working too. I put it into reader mode, and tried many times without success. My flipper zero can recognize the 2 badges. I just receive it and connect to my pc and phone withou issues.
Pls someone help!

GameTec-live, 2024-09-26:
did you read the troubleshooting docs

Jasperwolffe, 2024-10-17:
I'm in the same boat, I've had 0 success in getting it to read cards or tags. Have we had any success in getting it to function?

Lerold, 2024-10-17:
Does the RF Symbol flash when you read a tag with the Chameleon 
or if you try to read the Chameleon with the Flipper?

GameTec-live, 2024-10-17:
@Lerold thats intended behaviour, yes

Xample, 2024-10-25:
Hello, I have a Chameleon Ultra, which was reading LF but stopped recognizing right after upgrading to v2 (or even a nightly build). [I'm looking for a way to re-install the FW 1.0](https://github.com/RfidResearchGroup/ChameleonUltra/issues/236)

OpM3TA, 2025-06-27:
Currently experiencing this. I've already disassembled, cleaned with iso, I've tried using RF tape between the ant board as well, nada. Tried different fw versions and it just doesn't seem to read LF. :( a time or two it "glitched" or probably wasn't I na good enough spot idk, and. It "read" a tag but everything was 0.

Not sure what else to try or if I can modify with different ant

gtx495, 2025-07-16:
Expirienced the same problem with LF. Tried everything. But occasionally I've found the way it works :-)
If I place LF card to the side where the A nd B buttons are - everything works as expected.

OpM3TA, 2025-08-24:
> Expirienced the same problem with LF. Tried everything. But occasionally I've found the way it works :-)
> If I place LF card to the side where the A nd B buttons are - everything works as expected. 

Also worth mentioning this is what you're supposed to do lol.

But still unfortunately, no reading of LF for me, in particular HID rfid badges (125khz). Very frustrating as I do not really have a use for HF read/emu.

gtx495, 2025-08-24:
> Also worth mentioning this is what you're supposed to do lol.
>
> But still unfortunately, no reading of LF for me, in particular HID rfid badges (125khz). Very frustrating as I do not really have a [...]

by the way.. some LF cards are not readable till I'll rotate'em on 180degrees. The others work both sides.