RfidResearchGroup/ChameleonUltra #236: Downgrading to Basic Factory Firmware (1.0)

Xample, 2024-10-25:
Hello, I would like to downgrade to v1 (since v2 broke the LF reading functionality on my device). However, unlike the [v2 artifacts](https://github.com/RfidResearchGroup/ChameleonUltra/releases/tag/v2.0.0), the [v1 release](https://github.com/RfidResearchGroup/ChameleonUltra/releases/tag/v1.0) does not include an ultra-dfu-full.zip file.

I am using MTools BLE (on iPhone) to flash my Chameleon Ultra, but as far as I know, it can only load firmware in a packaged .zip format. Unfortunately, the v1 release only provides the app update in [chameleon_ultra_app_update.zip](https://github.com/RfidResearchGroup/ChameleonUltra/releases/download/v1.0/chameleon_ultra_app_update.zip), and there is no full firmware .zip package available.

I attempted to package the .hex file into a .zip using the nrfutil tool, but it seems to be more complicated than just packing the .hex with a key (possibly a manifest issue?).

I also tried to rebuild v1 using the v2 Docker builder, but the versions are too different, and I encountered compilation errors.

So, unless I’m completely off track, could someone assist me with the following:

Is there an ultra-dfu-full.zip available for v1?
How can I flash my device using the .hex file without an XXXLink (I’m on macOS)?

GameTec-live, 2024-10-25:
Chameleon Ultra GUI supports flashing .zip files. download the 1.0 zip file and flash that using CUGUI...
You should also be able to flash said zip via nrfutil. You only update the app, theres no need to flash a full package as the bootloader cant be updated anyways*.
(I also dont consider mtools trustworthy, but thats a different topic)

###### * Unless using SWD probes

Xample, 2024-10-25:
Thank your for the quick answer so just to clarify:
I’m using both CU GUI and MTools ble. I’m mostly using MTools ble for the update because CU GUI at first has no progress bar but overall, it never finishes the update, actually it did finish but only **one time** with the very first update on one of my 2 Chameleons.

As the full packages (when available in v2 or dev) provide both the app and the firmware packed, I therefore thought that both would be updated. 

So when you say : 
Chameleon Ultra GUI supports flashing .zip files. download the 1.0 zip file I guess do you mean this file : [chameleon_ultra_app_update.zip](https://github.com/RfidResearchGroup/ChameleonUltra/releases/download/v1.0/chameleon_ultra_app_update.zip) which should put me back to « firmware » 1 or more likely the **app v1**. I will try asap but If this is not the case I would be confused on how did the chameleon be upgraded to v2.

thank you

GameTec-live, 2024-10-25:
"CU GUI at first has no progress bar" yes it has? its at the bottom lol Very wierd that it doesnt finish the update though... might be a bug...
And yes, i do indeed mean the file you linked. afaik its also the only zip file provided in the 1.0 release...

Xample, 2024-10-26:
Hello, regarding the apps: here is screenshot on iOS of CU GUI
[image]
(it never finished the update)
while MTools BLE
[image]
Is working properly.

Anyway, I flashed my device using the [chameleon_ultra_app_update.zip](https://github.com/RfidResearchGroup/ChameleonUltra/releases/download/v1.0/chameleon_ultra_app_update.zip) The firware is now again recognized as v1 properly. And when a read a LF card unlike the v2 I get a result, but it is just a fake data. Actually, each time I try reading in LF, I get the same result (event if NO card is nearby). It simply means it's not a regression, just an improvement of the V2 which possibly reads something but is unable to decrypt the card.

In short: the V1 is just not working / reliable at all

GameTec-live, 2024-10-26:
well there is a progress bar... It just doesnt initialize the flashing process, so theres probably some bug somewhere...
[image]

And about your lf issues; idk... basically everyone seems to have lf issues altough 1.0 seems to work a bit better? I just rarely work with lf so it isnt a issue for me, but yeah, its more or less very broken...

Id say you should probably close this issue though as youve managed to successfully downgrade the fw

closed 2024-10-27 by Xample

whywilson, 2024-10-29:
Here's a web tool from @taichunmin to flash 1.0, 2.0 or latest dev firmware fastly.
https://taichunmin.idv.tw/chameleon-ultra.js/dfu.html

GameTec-live, 2024-10-29:
> Here's a web tool from @taichunmin to flash 1.0, 2.0 or latest dev firmware fastly.
> https://taichunmin.idv.tw/chameleon-ultra.js/dfu.html

right, that thing exists too, yep