RfidResearchGroup/ChameleonUltra #238: Issue with MIFARE Ultralight EV1 Emulation: Missing Data in Pages 18 and 19

Septimus4, 2024-11-12:
***Things to try before submitting a bug report***

- I have read the [troubleshooting guide](https://github.com/RfidResearchGroup/ChameleonUltra/blob/main/docs/troubleshooting.md)
- I checked existing [issues](https://github.com/RfidResearchGroup/ChameleonUltra/issues)
- I am using the latest firmware and CLI

**Describe the bug**  
When attempting to emulate a MIFARE Ultralight EV1 48-byte tag, the ChameleonUltra is missing data on pages 18 and 19. One of these pages contains the tag's password, which seems necessary for authentication. Emulating this tag with the ChameleonUltra fails to unlock the door, suggesting missing or incorrect data.

**To Reproduce**  
Steps to reproduce the behavior:
1. Use Proxmark to read the MIFARE Ultralight EV1 48-byte tag with the following command:
   ```bash
   [usb] pm3 --> hf mfu dump -k 8D322569 --ns
   ```
2. Attempt to read the same tag using the ChameleonUltra:
   ```bash
   [USB] chameleon --> hf mfu dump -k 8D322569
   ```
3. Compare the Proxmark and ChameleonUltra outputs. Notably, pages 18 and 19 on the Chameleon are populated with `00000000` instead of the expected values (`8D322569` in page 18 and `EC2F0000` in page 19).

**Expected behavior**  
The ChameleonUltra should accurately replicate all data pages of the MIFARE Ultralight tag, including pages 18 and 19, to enable successful emulation and authentication.

**Screenshots**  
**Proxmark Output:**
```
[usb] pm3 --> hf mfu dump -k 8D322569 --ns
[+] TYPE: MIFARE Ultralight EV1 48bytes (MF0UL1101)
[+] Reading tag memory...

[=] MFU dump file information
[=] -------------------------------------------------------------
[=] Version..... 00 04 03 01 01 00 0B 03
[=] TBD 0....... 00 00
[=] TBD 1....... 00
[=] Signature... 31 C9 18 7A A2 2F F4 F1 34 53 18 1D 7A BB 75 9B
[=]              5B 24 FC DC E9 01 84 A3 66 C4 28 26 51 61 5C F4
[=] Counter 0... 00 00 00
[=] Tearing 0... BD
[=] Counter 1... 00 00 00
[=] Tearing 1... BD
[=] Counter 2... 00 00 00
[=] Tearing 2... BD
[=] Max data page... 19 ( 80 bytes )
[=] Header size..... 56 bytes

[=] -------------------------------------------------------------
[=] block#   | data        |lck| ascii
[=] ---------+-------------+---+------
[=]   0/0x00 | 04 36 FA 40 |   | .6.@
[=]   1/0x01 | D2 52 1B 90 |   | .R..
[=]   2/0x02 | 0B 48 08 00 |   | .H..
[=]   3/0x03 | 9A 02 56 20 | 1 | ..V
[=]   4/0x04 | 19 6F 81 52 | 0 | .o.R
[=]   5/0x05 | 11 0B 73 48 | 0 | ..sH
[=]   6/0x06 | B6 76 E4 9B | 0 | .v..
[=]   7/0x07 | D2 2A 9F 61 | 0 | .*.a
[=]   8/0x08 | A8 88 E4 3A | 0 | ...:
[=]   9/0x09 | 5A 82 24 41 | 0 | Z.$A
[=]  10/0x0A | 00 00 00 00 | 0 | ....
[=]  11/0x0B | 00 00 00 00 | 0 | ....
[=]  12/0x0C | 00 00 00 00 | 0 | ....
[=]  13/0x0D | 00 00 00 00 | 0 | ....
[=]  14/0x0E | 00 00 00 00 | 0 | ....
[=]  15/0x0F | 00 00 00 00 | 0 | ....
[=]  16/0x10 | 00 00 00 10 | 0 | ....
[=]  17/0x11 | 80 05 00 00 | 0 | ....
[=]  18/0x12 | 8D 32 25 69 | 0 | .2%i
[=]  19/0x13 | EC 2F 00 00 | 0 | ./..
[=] ---------------------------------
```

**ChameleonUltra Output:**
```
[USB] chameleon --> hf mfu dump -k 8D322569
 - Detected tag type as Mifare Ultralight EV1 48b.
 - PACK: ec2f
 - Page  0: 0436fa40
 - Page  1: d2521b90
 - Page  2: 0b480800
 - Page  3: 9a025620
 - Page  4: 196f8152
 - Page  5: 110b7348
 - Page  6: b676e49b
 - Page  7: d22a9f61
 - Page  8: a888e43a
 - Page  9: 5a822441
 - Page 10: 00000000
 - Page 11: 00000000
 - Page 12: 00000000
 - Page 13: 00000000
 - Page 14: 00000000
 - Page 15: 00000000
 - Page 16: 00000010
 - Page 17: 80050000
 - Page 18: 00000000
 - Page 19: 00000000
```

**Host (please complete the following information):**  
- OS and version: Ubuntu 24.04.1 LTS  
- ChameleonUltra CLI hardware version:
  ```
  [USB] chameleon --> hw version
  - Chameleon Ultra, Version: v2.0 (v2.0.0-203-g8a797b8)
  ```

**Additional context**  
I am unsure if there’s an additional configuration step required or if this is a firmware issue. Any insights on how to address missing data for pages 18 and 19 would be appreciated. Thank you!

Septimus4, 2024-11-12:
I have a second issue but since it could be related I will add it here, and if it is not I will move it to a new one.

**Describe the bug**  
When attempting to use dumps from Proxmark as emulation data for the ChameleonUltra, I receive errors indicating that the dump file is too large for the available slot. I tried importing both `.bin` and `.json` formats, but both returned the same error message.

Is there a way to use proxmark dumps directly ?

**To Reproduce**  
Steps to reproduce the behavior:
1. Create a MIFARE Ultralight EV1 48-byte dump with Proxmark.
2. Attempt to load the Proxmark dump into the ChameleonUltra with the following commands:
   ```bash
   [USB] chameleon --> hf mfu eload -f hf-mfu-0436FAD2521B90-dump.bin
   Dump file is too large for the current slot (expected 80 bytes).
   ```
   And similarly:
   ```bash
   [USB] chameleon --> hf mfu eload -f hf-mfu-0436FAD2521B90-dump.json
   Dump file is too large for the current slot (expected 80 bytes).
   ```

**Expected behavior**  
The ChameleonUltra should accept the full Proxmark dump.

**Hex Dumps for Reference:**

Proxmark `.bin` Dump:
```
❯ hexdump -C hf-mfu-0436FAD2521B90-dump.bin
00000000  00 04 03 01 01 00 0b 03  00 00 00 13 31 c9 18 7a  |............1..z|
00000010  a2 2f f4 f1 34 53 18 1d  7a bb 75 9b 5b 24 fc dc  |./..4S..z.u.[$..|
00000020  e9 01 84 a3 66 c4 28 26  51 61 5c f4 00 00 00 bd  |....f.(&Qa\.....|
00000030  00 00 00 bd 00 00 00 bd  04 36 fa 40 d2 52 1b 90  |.........6.@.R..|
00000040  0b 48 08 00 9a 02 56 20  19 6f 81 52 11 0b 73 48  |.H....V .o.R..sH|
00000050  b6 76 e4 9b d2 2a 9f 61  a8 88 e4 3a 5a 82 24 41  |.v...*.a...:Z.$A|
00000060  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000070  00 00 00 00 00 00 00 00  00 00 00 10 80 05 00 00  |................|
00000080  8d 32 25 69 ec 2f 00 00                           |.2%i./..|
00000088
```

Chameleon Dump:
```
❯ hexdump -C chameleon

00000000  04 36 fa 40 d2 52 1b 90  0b 48 08 00 9a 02 56 20  |.6.@.R...H....V |
00000010  19 6f 81 52 11 0b 73 48  b6 76 e4 9b d2 2a 9f 61  |.o.R..sH.v...*.a|
00000020  a8 88 e4 3a 5a 82 24 41  00 00 00 00 00 00 00 00  |...:Z.$A........|
00000030  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000040  00 00 00 10 80 05 00 00  00 00 00 00 00 00 00 00  |................|
00000050
```

**Host (please complete the following information):**  
- OS and version: Ubuntu 24.04.1 LTS  
- ChameleonUltra CLI hardware version:
  ```
  [USB] chameleon --> hw version
  - Chameleon Ultra, Version: v2.0 (v2.0.0-203-g8a797b8)
  ```

GameTec-live, 2024-11-12:
is the slot configured as a 4k or 1k?

Septimus4, 2024-11-12:
I am not sure to understand, since it's a Ultralight Ev1 I used the those commands:
hw slot type -s 3 -t MF0UL11
hw slot init -s 3 -t MF0UL11

Septimus4, 2024-11-12:
After further testing, here’s an update:

1. I successfully wrote to an MF0UL21 tag, which provided more storage space. However, this did not unlock the door, and I’m not comfortable with this approach overall.

2. I then attempted to convert my Proxmark MIFARE Ultralight dump to the ChameleonUltra format. My understanding is that I should isolate only the card data blocks from the Proxmark dump and arrange them in a format the Chameleon will accept. This conversion involves extracting only the raw data from Proxmark’s blocks section, excluding any other metadata.

Here's a comparison of the hexdumps for reference:

```
❯ hexdump -C hf-mfu-0436FAD2521B90-handcrafted-chameleon.bin
00000000  04 36 fa 40 d2 52 1b 90  0b 48 08 00 9a 02 56 20  |.6.@.R...H....V |
00000010  19 6f 81 52 11 0b 73 48  b6 76 e4 9b d2 2a 9f 61  |.o.R..sH.v...*.a|
00000020  a8 88 e4 3a 5a 82 24 41  00 00 00 00 00 00 00 00  |...:Z.$A........|
00000030  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000040  00 00 00 10 80 05 00 00  8d 32 25 69 ec 2f 00 00  |.........2%i./..|
00000050
```

After preparing the file in this manner, I loaded it into the Chameleon without any errors or warnings using the following commands:

```
[USB] chameleon --> hw slot type -s 3 -t MF0UL11
 - Set slot 3 tag type success.
[USB] chameleon --> hw slot init -s 3 -t MF0UL11
 - Set slot tag data init success.
[USB] chameleon --> hf mfu eload -f hf-mfu-0436FAD2521B90-handcrafted-chameleon.bin
 - Ok
...
```

Though I had to set up the econfig values manually (which is quite tedious), the emulation finally worked, and the Chameleon successfully unlocked the door.

**Key takeaways and issues identified:**

1. **Potential bug** in the dump command for MIFARE Ultralight EV1 tags with password protection, unless I’m missing something.
2. **Incompatibility** between Proxmark and Chameleon dump formats.
3. **Manual econfig setup** for each use is cumbersome.

@GameTec-live, could you advise if the roadmap and documentation are up-to-date?

I’m open to contributing by exploring the possibility of adding Proxmark dump import support for the Chameleon. This could also automate econfig population from the imported dump. I could also investigate the incomplete block dump issue, though I’m relatively new to NFC/RFID and haven't worked on embedded systems in a long time, so I can’t guarantee significant contributions yet.

GameTec-live, 2024-11-12:
Well, first off, sorry, i totally missed that your talking about ultralight (ig i shouldnt respond to issues on my phone at 6am lmao).
Thanks for all of that info. Yes, the CU and pm3 format are incompatible. The CLI in general is quite barebones. (may i recommend ChameleonUltraGUI? XD)
The roadmap and docs should mostly be up to date and correct? So if you want to do some development, feel free to go ahead, especially as stuff like PM3 imports would be client side and thats just pretty easy to deal with python.

Septimus4, 2024-11-12:
No worries thank you for your help and informations.
I did try to use the Gui first but I could not dump the tag, and when trying to import the dump from proxmark it seemed to only allow me to select mifare classic types.
Since there was multiple underlying issues I should probably try again now that I have a valid dump to import.

GameTec-live, 2024-11-12:
in the gui imports are still missing mfu support, didnt have the time yet.
You should be able to read it though?

Septimus4, 2024-11-13:
I could read the tag uid, but not dump the content since it's password protected.
To dump I would need to provide the password
Or sniff the reader, and I think both are not supported

198sashko1, 2024-11-28:
Добрый вечер. Можно добавить пароль вручную. Но где его взять? Как его извлечь со считывателя?

GameTec-live, 2024-11-28:
@198sashko1 In English please

198sashko1, 2024-11-28:
Good evening. You can add a password manually. But where can I get it? How do I get it out of the reader?

Kirillko312, 2025-02-10:
> Good evening. You can add a password manually. But where can I get it? How do I get it out of the reader?

I join in the request. Is it possible to add automatic hex password search using a dictionary or some other way?

theonedemon, 2025-02-12:
hi, any news?

mindbreaker2, 2025-03-11:
You can use the proxmark to sniff the traffic between the ultralight and the reader. Then you'll be able to see the pwd (from my understanding)

198sashko1, 2025-03-11:
Это невозможно. Не имею проксиарк. Да и с ноутбуком у дома стоять не вариант.

mindbreaker2, 2025-03-11:
> Это невозможно. Не имею проксиарк. Да и с ноутбуком у дома стоять не вариант.

You need to keep it in English. 
If you don't have a proxmark you are basically screwed. You can try obvious passwords but their is a counter, with to many attempts you'll brick your card.

198sashko1, 2025-03-11:
It's my hobby. So I couldn't screw up. For the sake of one or two keys, buy a proxyark and stand at the entrance, next to the intercom? Such a hobby

198sashko1, 2025-03-11:
Moreover, the chameleon can theoretically sniff the reader itself. But the developers don't add it. And even if you have a proxmark, there are no instructions on how to do it all

GameTec-live, 2025-03-11:
> Moreover, the chameleon can theoretically sniff the reader itself. But the developers don't add it. And even if you have a proxmark, there are no instructions on how to do it all

No, the chameleon cant sniff the reader itself. (Whatever that may mean)
It cant sniff at all.

Also, "The developers"; Open up VS code, clone the repo and simsala bim, your a dev too that could implement whatever you like.

mindbreaker2, 2025-03-11:
> Moreover, the chameleon can theoretically sniff the reader itself. But the developers don't add it. And even if you have a proxmark, there are no instructions on how to do it all

There is a ton of documentation and tutorials, you just need to search. 
You can find posts on the proxmark forum (mirror) which explains exactly how to do it. Iceman also made a discord with pretty much all the knowledge you want and the people are willing to help if you ask politely. (english only since the proxmark community is from all over the world)

198sashko1, 2025-03-11:
> > Более того, хамелеон теоретически может сам обнюхивать считыватель. Но разработчики этого не добавляют. И даже если у вас есть проксмарк, нет никаких инструкций, как это все сделать
> 
> Нет, хамелеон не может учуять самого читателя. (Что бы это ни значило) Он вообще не может учуять.
> 
> Также «Разработчики»: откройте VS code, клонируйте репозиторий и симулируйте BIM, вы тоже разработчик, который может реализовать все, что угодно.

Unfortunately, I'm a complete zero at this. Therefore, it is unrealistic for me to contribute to this project myself.

198sashko1, 2025-03-11:
> > Более того, хамелеон теоретически может сам обнюхивать считыватель. Но разработчики этого не добавляют. И даже если у вас есть проксмарк, нет никаких инструкций, как это все сделать
> 
> Есть тонна документации и руководств, вам просто нужно поискать. Вы можете найти сообщения на форуме proxmark (зеркало), где подробно объясняется, как это сделать. Iceman также создал дискорд, где есть практически все необходимые вам знания, и люди готовы помочь, если вы вежливо попросите. (только на английском, так как сообщество proxmark состоит из людей со всего мира)

The ice man is the guru of NSF. But I'm afraid I won't be able to do it. I work with intercoms. And I can't imagine doing it on the street. With a laptop and a proxmark. The police will take it away.

198sashko1, 2025-03-11:
I have great respect for your group. And by no means do I consider thieves. All this is an incorrect auto-translation. I meant that standing with a laptop and a proxmark would attract the attention of the police. And I have to explain for a long time that I work for an intercom company and am testing an intercom for hacking.

198sashko1, 2025-03-11:
It would be easier with a chameleon and a phone.

GameTec-live, 2025-03-11:
> I have great respect for your group. And by no means do I consider thieves. All this is an incorrect auto-translation. I meant that standing with a laptop and a proxmark would attract the attention [...]

If its your system and you have a intercom company badge and you have a right to be there, theres no need to be afraid of the police or have any long discussions with them.
But as said, feel free to contribute, everythings opensource, rn it isnt possible tho

198sashko1, 2025-03-11:
> > Я с большим уважением отношусь к вашей группе. И ни в коем случае не считаю вас ворами. Все это некорректный автоперевод. Я имел в виду, что стоять с ноутбуком и проксмарком привлечет внимание полиции. И мне придется долго объяснять, что я работаю в домофонной компании и тестирую домофон на взлом.
> 
> Если это ваша система и у вас есть значок компании-интеркома, и вы имеете право там находиться, нет нужды бояться полиции или вести с ними долгие дискуссии. Но, как уже было сказано, не стесняйтесь вносить свой вклад, все с открытым исходным кодом, хотя это невозможно.

I'm not shy. I don't know how to help, where to start. If you had given instructions on what to test, remove the logs. I would love to make and publish it here. But, I repeat, I'm completely at a loss to come up with something and add it to the code. Only according to the instructions. I can't step aside.

Uhrien, 2025-04-07:
Hey there, I would also like to point out that it would be interesting to also properly emulate the correct value of the counters :D

GameTec-live, 2025-04-07:
@Uhrien should be doable

Uhrien, 2025-04-07:
Do you know by any chance how to flash some kind of DEV firmware on the chameleon ultra?
I'd like to contribute to the project, but I just don't know how to flash a firmware which is not the *latest official release* 
Any help would be extremely appreciated <3

WernBe220007, 2025-04-07:
see the docs. either use nrfutil or use the zip flashing function in cugui (hidden in the device settings)

GameTec-live, 2025-04-07:
see the docs. either use nrfutil or use the zip flashing function in cugui (hidden in the device settings)