RfidResearchGroup/ChameleonUltra #241: Dead project?

SoLongSidekick, 2024-12-30:
Is this a dead project? I'm only asking because the roadmap seems to not have been touched in a year or two.

GameTec-live, 2024-12-30:
This is a useless issue.
If you want to see what people are talking about and why there's slow progress, look at the various discords, but don't open a pointless issue.

panayotoff, 2025-01-20:
> This is a useless issue. If you want to see what people are talking about and why there's slow progress, look at the various discords, but don't open a pointless issue.

HI man, first of all, I **really** respect what you're doing. Just people come here, in Github to see progress and the latest stable release was over 20 months ago, and still quite a lot of things are missing or buggy ( expectations set from the initial campaign ). People are wondering is the project abandoned, that's all. I agree this particular ticket is useless, but the work on chameleon should be much better communicated IMO.

Kyhze, 2025-02-01:
@GameTec-live Do you think there is a chance the development will actively resume? It'd be sad to lose this beautiful project, imo. It has so much potential! The hardware is incredible. I love my little CU.

Semper-Viventem, 2025-02-01:
@Kyhze this is an open source project, people usually build driven by enthusiasm. All are welcome to contribute

Kyhze, 2025-02-01:
> [@Kyhze](https://github.com/Kyhze) this is an open source project, people usually build driven by enthusiasm. All are welcome to contribute

Indeed it is. I'm merely expressing my concern that the enthusiasm may be dying off. Which would be sad. I would happily contribute if I had the skills necessary, but I unfortunately do not.

brianbgarber, 2025-02-01:
Same here. Thank you @Kyhze  for bringing this topic up. I bought the CU because of the potential uses advertised at the get go... but understanding that there are risks in investing in this type of project. I know it's open source... but unfortunately I too lack the skills to develop software (I'm a hardware engineer and that work is already done!) and at this point, unfortunately, my CU is sitting in a drawer 

I don't follow (on discord or any other platform) or understand the dynamamics within the community that started this project but I too am sad that this project has stalled.  Personally I don't see this as a pointless issue since others are following and are wondering the same thing.

Skyler84, 2025-02-10:
Hi, new to the Chameleon. Hoping to contribute to the project in the coming weeks/months. Already working on Hitag2 emulation and refactoring some of the emulation code to make it a bit nicer to work with, and then likely to add a few more tag types (including Mifare Desfire?)
The Chameleon hardware seems really good and has a lot of potential, shame it hasn't taken off yet but hopefully with a few new features more people will be interested in the hardware and the software will get a bit more TLC

closed 2025-04-05 by GameTec-live