RfidResearchGroup/ChameleonUltra #244: [idea]Add a relay function to the Chameleon.

larblue, 2025-02-14:
I have an idea. Since the Chameleon supports Bluetooth communication and can also simulate LF and HF cards, is it possible to develop a feature that enables peer - to - peer (P2P) communication between two Chameleons?
Here's how it could work: Chameleon A simulates a card to interact with a card reader. It then sends the interaction information to Chameleon B via Bluetooth. Chameleon B, in turn, interacts with the real card. This way, remote card reading and writing can be achieved, which would be extremely useful.

GameTec-live, 2025-02-14:
sounds like an idea, feel free to try and implement it!

GibbilyGooo, 2025-02-15:
> [larblue] larblue created an issue [(RfidResearchGroup/ChameleonUltra#244)](https://github.com/RfidResearchGroup/ChameleonUltra/issues/244)

It’s junk! !

Sent from [Proton Mail](https://proton.me/mail/home) for iOS

Kyhze, 2025-02-15:
> It’s junk! !
> 
> Sent from [Proton Mail](https://proton.me/mail/home) for iOS
> […](#)

This isn't Facebook.

Nightingale0504, 2025-08-13:
but there are some problems: first, the effective distance of ble is just 10m, if you want to remote read/write card, this can;t be very useful.
if you want, we need mobile phones and network connections to accomplish long distance data transmition