RfidResearchGroup/ChameleonUltra #249: Do you guys think...?

realdaveblanch, 2025-04-03:
Please keep in mind we are still at the infancy of the project and a lot has still to be done.

Have first a look at the [Whitepaper](https://github.com/RfidResearchGroup/ChameleonUltra/blob/main/docs/technical_whitepaper.md), the Wiki [Roadmap](https://github.com/RfidResearchGroup/ChameleonUltra/wiki/Public-Roadmap) and [Wishlist](https://github.com/RfidResearchGroup/ChameleonUltra/wiki/Wishlist).  
No need for creating issues for well known missing supports, we are already aware :)

I have a SeedStudio XIAO nr52840, do you guys think that it could be possible to port the actual project to it? 
I received my chaamleon ultra yesterday and is awesome, and just found out that is the same chip as my Xiao. 
So, im guessing that if i solder a nfc antenna and edit the source i might be able to make it run this ROM.

Do you guys think is possible?

Thanks in advance (ill check the actual code and try to PR stuff ;))

**Is your feature request related to a problem? Please describe.**
Nope

**Describe the solution you'd like**
Nope

**Describe alternatives you've considered**
Nope

**Additional context**
Nope

GameTec-live, 2025-04-03:
No, it's not possible without a few extra modules.
But if you have these extra modules, it should just run fine.

realdaveblanch, 2025-04-05:
> No, it's not possible without a few extra modules.
> But if you have these extra modules, it should just run fine.

I just reviewed the hardware stuff and idk if I'm correct:

I only need more modules in case I want to use the LF stuff?? 

I mean, if I fork it and adapt the code for the standalone SEED XIAO nr52840 sense it may work for HF tags? 

Thanks you

GameTec-live, 2025-04-05:
Oh yeah, if your just trying to build a Chameleon Lite, that would work. Why would you tho?
Not to discourage, it sounds like a cool project, but CU clones are like 20€ from China and they already contain the extra chips for LF, etc. The XIAO modules are typically 15 - 20€, so there isnt a huge difference and your missing like half the functionality.
Anyways, im going to close this issue, as this isnt really a CU issue. Happy tinkering!

closed 2025-04-05 by GameTec-live