RfidResearchGroup/ChameleonUltra #253: Mifare Ultralight EV1 Emulation Fails/Crashes Readers (Flipper Zero) Due to Incorrect Command Handling and Timing Issues

Uhrien, 2025-04-13:
Here are the steps I've gone through for dumping a Mifare Ultralight EV1 with the CU and save the correct dump inside the CU, slot 1:
hw connect
hf mfu dump -f ul_ev1_original2.bin -t bin
hw slot delete -s 1 --hf
hw slot type -s 1 -t MF0UL21
hw slot change -s 1
hf mfu eload -f ul_ev1_original2.bin -t bin
hf mfu ewcnt -c 0 -v 16645889
hf mfu ewcnt -c 1 -v 0
hf mfu ewcnt -c 2 -v 512
hf mfu econfig --uid 04986912A11994
hf mfu econfig --set-version 0004030101000E03
hf mfu econfig --set-signature 7773B41E9CEDD0CCC444F60885446D078DCC2F65D9BB3010435E84BFA2DE951D
hw slot store
hw slot enable -s 1 --hf
hf mfu eview
hf mfu ercnt -c 0
hf mfu ercnt -c 1
hf mfu ercnt -c 2
hf mfu econfig --delete-ats
hw slot store

The Chameleon Ultra's Mifare Ultralight EV1 emulation causes issues with various readers: Flipper Zero crashes, Proxmark3 (hf mfu dump --ns) often fails on the first attempt, and some readers fail detection entirely.

Proxmark3 sniff logs reveal two main issues:

Incorrect response to unsupported commands: The Chameleon replies incorrectly to the Mifare Classic AUTH-A (0x60) command, causing reader instability (likely the Flipper crash). A genuine card remains silent.

Timing discrepancies: The Chameleon responds slightly faster during initial selection than a genuine card, potentially causing detection failures with stricter readers.

Steps to Reproduce:

Emulate a Mifare Ultralight EV1 on Chameleon Ultra.

Attempt to read with a Flipper Zero (observe crash).

Attempt to read with Proxmark3 (hf mfu dump --ns) (observe potential first-try failure).

Sniff the interaction using Proxmark3 (hf 14a sniff) while reading with Flipper Zero.

Evidence / Logs:

Key difference observed when Flipper Zero sends Classic AUTH-A (60 00 F5 7B):

(1) Chameleon Ultra (Emulation) Log Excerpt:

Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation
------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------
... (Selection OK) ...
    2815584 |    2820288 | Rdr |60  00  F5  7B                                                           |  ok | !!! AUTH-A(0) Command (Classic) !!!
    2821524 |    2833172 | Tag |00  04  03  01  01  00  0E  03  45  89                                   |  ok | !!! INCORRECT RESPONSE from Chameleon !!!
  161106144 |  161107136 | Rdr |52(7)                                                                    |     | WUPA <-- Flipper likely crashed/restarted
Use code [with caution](https://support.google.com/legal/answer/13505487).
(2) Genuine Ultralight EV1 Card Log Excerpt:

Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation
------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------
... (Selection OK) ...
    2662416 |    2667120 | Rdr |60  00  F5  7B                                                           |  ok | !!! AUTH-A(0) Command (Classic) !!!
    2991760 |    2992752 | Rdr |52(7)                                                                    |     | WUPA <-- *** NO RESPONSE FROM GENUINE CARD (Correct) ***
Use code [with caution](https://support.google.com/legal/answer/13505487).
Analysis Summary:

Critical: Chameleon incorrectly replies to Classic AUTH-A 0x60, likely sending GET_VERSION data instead of remaining silent. This crashes the Flipper.

Secondary: Chameleon's initial selection responses (ATQA, UID, SAK) are consistently ~7-9µs faster than the genuine card, potentially violating reader FDT expectations and causing detection failures.

Suggested Fixes:

Fix Unsupported Command Handling: Modify firmware to ensure the Chameleon does not respond to commands unsupported by Ultralight EV1 (especially Classic AUTH-A 0x60). It must remain silent. Distinguish clearly between Classic 0x60 and EV1 GET_VERSION 0x60.

Adjust Initial Timings: Introduce small delays before ATQA/UID/SAK transmission to better match genuine card timing and ISO 14443-3 FDT specifications.

Audit EV1 Commands: Verify the implementation of all EV1-specific commands (GET_VERSION, READ_SIG, READ_CNT, PWD_AUTH, etc.) against the NXP datasheet for accuracy.

Environment:

Reader: Flipper Zero (latest official fw)

Sniffer: Proxmark3 Easy (latest RRG/Iceman fw)

Device: Chameleon Ultra (latest main branch fw)

GameTec-live, 2025-04-13:
Thanks for providing more detail on this issue.
Ig it might be worth opening a issue on the flipper side too, as just sending wrong data shouldnt crash the entire device?

Uhrien, 2025-04-13:
I think so, but I don't know if there's also an issue with the timing window during the initial handshake (and that may also have some influence of the flipper zero's crash thing)

xc52194xc, 2025-04-20:
这个出现了类似的情况，9-10 区 模拟后  有时候能正确识别，但是多数直接假死，导致奔溃，跑飞，切换其它数据也无效，没有了模拟功能，需要重新启动连接才能正常，出现这个BUG后会导致他不发射NFC信号

GameTec-live, 2025-04-20:
Can we please keep this thread in english?

xc52194xc, 2025-04-20:
> 这个出现了类似的情况，9-10 区 模拟后 有时候能正确识别，但是多数直接假死，导致奔溃，跑飞，切换其它数据也无效，没有了模拟功能，需要重新启动连接才能正常，出现这个BUG后会导致他不发射NFC信号

This has a similar situation, after the 9-10 area is simulated, sometimes it can be correctly identified, but most of them are directly faked, resulting in crashing, running away, switching other data is also invalid, there is no simulation function, you need to restart the connection to normal, after this BUG will cause him not to send NFC signals

xc52194xc, 2025-04-20:
> 我们能把这个帖子保持为英文吗？

This has a similar situation, after the 9-10 area is simulated, sometimes it can be correctly identified, but most of them are directly faked, resulting in crashing, running away, switching other data is also invalid, there is no simulation function, you need to restart the connection to normal, after this BUG will cause him not to send NFC signals

xc52194xc, 2025-04-20:
> 我们能把这个帖子保持为英文吗？
The normal upload simulation is successful, the first access control is normal, the brush elevator is all invalid, the restart can be normal, the previous one also failed, this is the video and data
https://github.com/user-attachments/assets/e0baa820-ad3c-4f14-89f2-cb267c5ec64a

0 
45 39 DC F4 54 08 04 00 02 80 89 C7 36 C7 DB 1D
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
FF FF FF FF FF FF FF 07 80 69 FF FF FF FF FF FF

1 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
FF FF FF FF FF FF FF 07 80 69 FF FF FF FF FF FF

2 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
FF FF FF FF FF FF FF 07 80 69 FF FF FF FF FF FF

3 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
FF FF FF FF FF FF FF 07 80 69 FF FF FF FF FF FF

4 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
FF FF FF FF FF FF FF 07 80 69 FF FF FF FF FF FF

5 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
FF FF FF FF FF FF FF 07 80 69 FF FF FF FF FF FF

6 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
FF FF FF FF FF FF FF 07 80 69 FF FF FF FF FF FF

7 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
A0 F3 C3 0C 0C 0C 0C 0C 0C 0C D7 9A F3 F3 F3 F3
F3 F3 F3 F3 F3 F3 F3 F3 F3 F3 F3 F3 F3 F3 F3 F3
0B 5F B4 CF 2F D5 FF 07 80 00 0B 5F B4 CF 2F D5

8 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
03 03 03 03 03 03 03 03 FC FC FC FC FC FC FC FC
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0B 5F B4 CF 2F D5 FF 07 80 00 0B 5F B4 CF 2F D5

9 
00 00 50 E0 FF FF AF 1F 00 00 50 E0 24 DB 24 DB
4A 54 91 7B 4A D1 7B 39 8B 2E 2E 7B 2E 7B 7B 7B
7A 2A FB 84 7B 7B 7B 7B 7B 7B 7B 7B 7B 2E 25 97
F2 84 C8 E3 B2 07 FF 07 80 00 F2 84 C8 E3 B2 07

10 
9A 7B 0A C7 A3 04 84 84 84 84 84 84 84 84 84 84
84 84 84 84 84 7B 34 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
F2 84 C8 E3 B2 07 FF 07 80 00 F2 84 C8 E3 B2 07

11 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
F2 84 C8 E3 B2 07 FF 07 80 00 F2 84 C8 E3 B2 07

12 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
F2 84 C8 E3 B2 07 FF 07 80 00 F2 84 C8 E3 B2 07

13 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
F2 84 C8 E3 B2 07 FF 07 80 00 F2 84 C8 E3 B2 07

14 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 50 E0 FF FF AF 1F 00 00 50 E0 3A C5 3A C5
F2 84 C8 E3 B2 07 FF 07 80 00 F2 84 C8 E3 B2 07

15 
50 00 42 11 28 16 00 FF FF FF FF FF FF FF FF FF
FF 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 FF FF FF FF 00 00 00 00 3E C1 3E C1
01 02 03 04 05 06 FF 07 80 69 01 02 03 04 05 06

turbocool3r, 2025-04-27:
If you check the [datasheet](https://www.nxp.com/docs/en/data-sheet/MF0ULX1.pdf) (section 10.1), there is the GET_VERSION command with ID 0x60. Are you sure that the card you've used in your test is a genuine Ultralight EV1?

Uhrien, 2025-04-30:
Yup, at least that's what I'm getting from my Proxmark3 & Flipper Zero's readings.
Furthermore, my Flipper Zero is emulating the whole card + counters flawlessly, but I still can't manage to perform a full proper emulation with CU's firmware