RfidResearchGroup/ChameleonUltra #255: 125кГц еммарин к домафону элтис с Хамемелеона не открывает дверь

Aleks1958, 2025-04-17:
Hello, question: Simple 125 kHz emmarin to the Eltis Chamemeleon intercom reacts like this: a white diode lights up on the side and shows the slot in which the key is blue, if you hold it, it will continue to shine white, but does not give grounds that Emmarin is sure exactly, the Speaker made keys from the intercom and magnetic cards TMRF-2RW, he produced chemical rewritable ones for this Eltis intercom, a tablet, wrote in chamemeleon but it opens, tell me, what could it be, who knows? Thanks in advance

GameTec-live, 2025-04-17:
can you change the title to english please?

Aleks1958, 2025-04-17:
Hello, question: Simple 125 kHz emmarin to intercom Eltis Chamemeleon Ultra reacts like this: white diode lights up on the side and shows blue slot where key is, if you hold it it will continue to shine white, but does not open door of intercom Eltis, that emmarin 125 kHz is for sure, did Duplicate keys from intercom and magnetic cards TMRF-2RW, it recorded rewritable keys to this intercom Eltis, this blank was registered in chamemeleon but it does not open door, blanks open and with Chamemeleon Ultra it does not work, tell me what could be? Thanks in advance.

GameTec-live, 2025-04-17:
Are you sure the card your using is EM410X? The CU rn only supports EM410X on the LF side.

xutpyh, 2025-07-19:
Hello! The same problem as the alex1958 with emulation of the EM410X 125 kHz Chameleon Ultra. The Eltis intercom works with the original Vizit Safe Home key and a clone on the T55xx recorded Chemleon. When Chameleon emulates the same key, the Eltis intercom does not work (the white LED RF lights up on Chameleon, but nothing happens).

[image]
[image]
[image]

xutpyh, 2025-07-19:
1 screenshot - the original key - works.
2 screenshot - a clone of the original key - works.
3 screenshot - emulation of Chameleon Ultra of the original key - does not work.

xutpyh, 2025-07-22:
I suppose that in the intercom Eltis, some filters are used as protection against emulation.

Savagesin, 2025-08-08:
I also encountered a problem.
Chameleon Ultra.
Chameleon reads code from the Em410, but it does it incorrectly. 
It emulates a completely different code.
<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/U_124/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/U_124/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

00 | 67 | 68 | 00 | DB | D6 | 03 | 00 | em410 in Chameleon
-- | -- | -- | -- | -- | -- | -- | -- | --
00000000 | 01100111 | 01101000 | 00000000 | 11011011 | 11010110 | 00000011 | 00000000
1F | 00 | 68 | 0 | DB | D6 | 03 | 01 | Real data in device
00011111 | 00000000 | 01101000 | 00000000 | 11011011 | 11010110 | 00000011 | 00000001
6E | 00 | DE | AD | BE | EF | 00 | 01 | emul. Chameleon in device
01101110 | 00000000 | 11011110 | 10101101 | 10111110 | 11101111 | 00000000 | 00000001

</body>

</html>

Foxushka, 2025-08-10:
@Savagesin you performed that test on latest firmware, aka with hid prox support?

xutpyh, 2025-08-10:
> 

I have this problem too.

Savagesin, 2025-08-12:
> [@Savagesin](https://github.com/Savagesin)Вы проводили этот тест на последней прошивке, то есть с поддержкой HID-Prox?

Yes. Chameleon Ultra GUI offered firmware. he downloaded it and installed it. after restarting the device, I received this data.

Andrysky, 2025-09-04:
It seems that the problem has been fixed in the latest firmware (v2.1.0)

Aleks1958, 2025-09-04:
Yes, the problem is gone, everything works with firmware 2.1.0

closed 2025-09-04 by Aleks1958