RfidResearchGroup/ChameleonUltra #271: LF tag read issue

dmitsk, 2025-08-08:
**Firmware**
The latest UI(from play market) and firmware (2.0.0-247-g098e0a9)

**Describe the bug**
After success saving the card from read operation, you will see something xx xx 0000000000
Should be xx xx xx xx xx, no zeros at the end 

**To Reproduce**
Read and save LF tag EM410X
See the UUID

**

[image]

**

**Notes**
I tried with v2.0.0 firmware and there is no such issue, so possible the issue not in UI

Foxushka, 2025-08-09:
@TeCHiScy I've already fixed it at https://github.com/GameTec-live/ChameleonUltraGUI/pull/615/commits/1528f7a95a4e92c90e8785f75d9669199df3b164, please do not change anything again

TeCHiScy, 2025-08-09:
> [@TeCHiScy](https://github.com/TeCHiScy) I've already fixed it at [GameTec-live/ChameleonUltraGUI@1528f7a](https://github.com/GameTec-live/ChameleonUltraGUI/commit/1528f7a95a4e92c90e8785f75d9669199df3b164), please do not change anything again

Ok, thank you. Then I'd leave it not changed.

OArkh, 2025-08-11:
Please consider that not only saving of the card is broken.
Also read LF card in reader mode returns too many data bytes - at least too many for EM4100.

[image]

Foxushka, 2025-08-11:
@OArkh yes, known and already fixed at https://github.com/GameTec-live/ChameleonUltraGUI/pull/627 with extra changes in https://github.com/GameTec-live/ChameleonUltraGUI/pull/628