RfidResearchGroup/ChameleonUltra #286: LF Card not working with mixed technology reader

ushiocheng, 2025-08-28:
I was cloning a Schlage 9691T fob which is HID prox + Mifare classic. The mfc part worked but the hid prox part didn't. I tried to double check the reader is lf/prox so I put a field detecter to it. It shows as alternating between hf and lf which is presumably what confused the CU. Have anyone else had similar experiences?

- The reader is https://www.brivo.com/acs100/
- I tried to use a slot only for lf, didn't work
- pm3 reads emulated card fine, both hf and lf
- I copy the hid data onto t55xx and that worked for that reader
- My end solution is just buy a dual tech magic card and copy the data to it

ushiocheng, 2025-08-29:
Correction: If I put it in a slot only for lf and turn CU on before holding it to the reader, it works most of the time.

LupusE, 2026-02-02:
Could you check if this fixed your issue?
https://github.com/RfidResearchGroup/ChameleonUltra/commit/055ae2322ebf927f23eec61df8ef7b3bcba9e1dd#diff-f0fad6ea234de35166f1bf78bc4aa2a9bf86c567e267ca560805477bf75fb48dR74

Here the CU get a clean reset instead of a strange state. Sounds like what you observed.

mfcarroll, 2026-11-09:
@ushiocheng if you're still using this reader and interested in trying to get it to work better still let me know. The fix @LupusE mentioned is likely part of it, but it's also possible more can be done to align with a pulsed LF field, which just came up in #532 with a very picky, pulsed Paradox reader (Paradox and HID use a similar emulation type).