RfidResearchGroup/ChameleonUltra #293: [idea] MIFARE DESFire EV3 8k support

Archer7x, 2025-09-06:
Hi,
I'm dont know how to implement **MIFARE DESFire EV3 8k** support so I hope someone be able to implement support for **MIFARE DESFire EV3 8k** 

Thank you!

GameTec-live, 2025-09-06:
While technically possible, youd still need to own the system and have the keys. Soo use is limited.

X-Ryl669, 2025-09-09:
No, the keys are set by the user of the tag (usually) and are all different. You can brute force them, if you can survive for more than the duration of the universe.

Archer7x, 2025-09-09:
So there is no possibility that they use the standard keys. Like someone who never changed his WiFi password? Are these cards therefore uncloneable? If I want to have the most secure RFID setup should I use this card?

GameTec-live, 2025-09-09:
DESFire is typically considered uncloneable, yep. And yes, desfire and javacards (or smartcards in general) are more or less the only secure tech.