RfidResearchGroup/ChameleonUltra #301: Client shows wrong commit hash for the firmware

Steve707-AE, 2025-09-24:
**Describe the bug**
I'm on the last version of the firmware and the client shows  
`- Chameleon Ultra, Version: v2.1 (v2.1.0-7-g647631f)`
but I think it the hash should be `8998621` as listed in the page development release built from commit.

**To Reproduce**
Steps to reproduce the behavior.  
E.g.
1. Connect CU
2. Execute `hw version`

**Expected behavior**
Shows the correct hash of the commit.

**Host (please complete the following information):**
 - Windows

Steve707-AE, 2025-09-25:
Yes, I got the popup. 

[image]
Anyway, the commit `647631f` it's internal, so I think the commit hash is written wrong inside the firmware and both cli client and cu gui shows the wrong commit.

Foxushka, 2025-09-27:
https://github.com/RfidResearchGroup/ChameleonUltra/commit/647631f this is 647631f commit 
I've fixed firmware update checks in latest releases of CU GUI, maybe you are using old version (for example 1.3 release instead of main branch) and it downloaded wrong update. If you can reproduce it on latest version of CU GUI (https://github.com/GameTec-live/ChameleonUltraGUI/releases/download/dev/windows-installer.zip) create issue on our Github page

Steve707-AE, 2025-10-01:
The issue is not related only to CU GUI otherwise I would have opened an issue con that. If I check the version through the CLI it's also wrong.

[image]

So I think the bug is related to how the hash of the commit is written on the firmware.

[image]

The hash shouldn't be `8998621`?

Steve707-AE, 2025-10-05:
Moved to [https://github.com/GameTec-live/ChameleonUltraGUI/issues/750](url)

closed 2025-10-05 by Steve707-AE